alexliniger/MPCC #2: Real-Time Implementation

PranjaLBiswas27, 2019-01-31:
Hi,

Is the solver used here same as the one used within FORCES PRO. If not, does it run at the same speed as mentioned in the paper which is referred to in this repository?

Kind Regards
Pranjal Biswas

alexliniger, 2019-01-31:
No, it is a different solver, however, it is also a tailored real-time MPC solver. Regarding computation times, in the tests I did I observed that HPIPM is actually faster than FORCES Pro (at least the version I tested).
One thing I have to say is that the HPIPM Python Interface is slow. Thus, the simulation runs quite slow. If you want to solve it in real-time the whole code should be written in C++. (If there is enough interest I may also open-source a C++ version of the code.)

MTDzi, 2019-01-31:
Hi,

I would be very much interested in a C++ version, even a simplified one :+1:

PranjaLBiswas27, 2019-01-31:
Hi, yes a  C++ version  will be much appreciated. Also thanks for clearing out my query.

closed 2019-01-31 by PranjaLBiswas27

PranjaLBiswas27, 2019-04-18:
Hi @alexliniger,

Will you be making the C++ version of the code open-source anytime soon in the future? Please let me know.

Kind Regards
Pranjal Biswas