alexliniger/MPCC #6: Stability

PranjaLBiswas27, 2019-02-11:
Hi,

I read the paper referred to in this repository. In the video I see cars making turns at high speeds without spinning out. Also if I am correct the states of your system are (v_x, v_y, omega). So do you put any specific constraint on the states to guarantee stability at the limits of handling for the MPCC approach, or stability is not guaranteed? Looking forward to hear from you. 

Kind Regards
Pranjal Biswas

alexliniger, 2019-02-11:
No there are no constraints added for stability. The constraints only limiting the car to stay within the track, but other state constraints are not imposed. However, the car needs to fulfill the state dynamics which implicitly results in the car not spinning. One cost I sometimes used to further penalize too aggressive driving was a cost on the slip angle (something we used on the AMZ formula student car).

From a theoretical standpoint, stability could be guaranteed by imposing terminal constraints and terminal costs in the MPC problem. However, since the problem is nonlinear it is very hard to come up with terminal costs and constraints that guarantee recursive feasibility and stability. Also, stability may not be the concept you are looking for since this would mean the car wants to stand still in the future. Recursive feasibility which can be guaranteed using terminal constraints, on the other hand, is in my opinion very important, and I studied it during my PhD for a slightly different formulation (see https://arxiv.org/abs/1701.08735, and https://youtu.be/2BV0tG0Cd2g)

Best,
Alex

PranjaLBiswas27, 2019-02-11:
Is there a paper published on the work you did on AMZ student car regarding the same?

Also, correct me if I am wrong cause when you say cost on slip angle, I assume this makes the optimization problem non-convex. I am saying this without doing any analysis as such, but from the fact that if a state space including the slip angle as the state is considered, a general assumption on constant longitudinal speed across the prediction horizon is taken to keep the optimization problem convex. Else including both longitudinal velocity and slip angle as the states makes the optimization problem non-convex. So if my inference here is correct how do you deal with a non-convex optimization problem?

Also, I was trying to refer to stability itself, in the sense as done by Funke et al. in "Collision Avoidance Up to the Handling Limits for Autonomous Vehicles", where they define a stability envelope. Thus guaranteeing stability within the invariant set, while nothing can be said outside that. So I thought you might have somehow incorporated something similar since the cars are very stable at high speeds around the corners.

Do let me know your views.

Kind Regards
Pranjal Biswas

alexliniger, 2019-02-11:
There are two papers coming including MPCC for the AMZ car, both should be soon finished. 

The problem is anyhow a non-convex (nonlinear dynamics, non-convex cost, non-convex state constraints), and I solve it using a sequential quadratic programming approach (in the limit this is just one QP solve, also known as real-time iteration). Therefore, the nonlinear slip angle cost can just be linearized and resulting in a quadratic cost function. 

You could include these constraints, even though they do not prove stability or recursive feasibility in the given formulation, still, they are interesting a could help the performance. The issue is that by definition, first, that stability requires the state to go to an equilibrium which is only given if the car is standing still and second, that they do not guarantee that the track constraints are invariant. For this controller, we rely on the popular heuristic of using a long horizon the deal with this issue, which implicitly excludes points outside the safety envelope (at least in practice). 

However, including these constraints could be interesting, and after a quick look through the PhD thesis of J. Funke it should be possible, at least if his safety envelop is used. This would, however, neglect that different tire models are used.

PranjaLBiswas27, 2019-02-11:
Ahh, I did see the optimization problem is non-convex and a you use a LTV approximation of the non-linear dynamics and approximated quad. costs to solve a convex problem. But since I haven't had much experience with LTV MPC it didn't come to my mind that I can do the same with side slip angle. But in that case if I am correct, is it that we need to augment the state space with side slip angle dynamics as well along with states which have already been considered?

Well I am curious about the MPCC paper on the AMZ car, hope its done soon.

alexliniger, 2019-02-12:
It is not necessary to add an additional state, there are two ways to add a cost on the slip angle, both start with the cost which is equal to J_beta = beta^2 = atan(vy/vx)^2
1) you can take that cost and compute the second order Tyler series expansion
2) you can approximate beta using a first order Tylor expansion beta ~ A_beta [vx;vy] + b_beta, then beta^2 ~ [vx;vy]^T A_beta^T A_beta [vx;vy] + 2 b_beta A_beta [vx,vy] + b_beta^2
Both approaches lead to a quadratic cost which can be included in the QP.

PranjaLBiswas27, 2019-02-12:
Ahh, thanks a lot for the help here, really appreciate it.

closed 2019-02-12 by PranjaLBiswas27