alexliniger/MPCC #8: Cost Function of MPCC

PranjaLBiswas27, 2019-03-01:
Hi @alexliniger,

While going through your paper referenced in this repository, I see that in Equation 13a, in the cost function you have change in inputs (delta u_k) and change in velocity along the path (delta v_k). Please correct me if I am wrong, but as far as my understanding of MPC goes variables in cost function can only be terms related to the states and the inputs of the system, while delta u_k or  delta v_k are neither states nor the inputs of the system (Equation 12b and 13b). So how do these two terms in the cost function fit within the optimization framework here? 

Kind Regards
Pranjal Biswas

alexliniger, 2019-03-01:
Including differences of optimization variables into an optimization problem is generally no issue. However, some QP solvers mainly those optimized for MPC do not allow for that. 
In the paper and the implementation, we do this by actually lifting the state space and introducing an additional optimization variable which is the change of inputs. You can see this in the code in the comment here
https://github.com/alexliniger/MPCC/blob/master/Matlab/getMPCmatrices.m#L30
and implemented here 
https://github.com/alexliniger/MPCC/blob/master/Matlab/getMPCmatrices.m#L224

PranjaLBiswas27, 2019-03-04:
Thanks, I will try to write down the optimization problem step by step to understand it in that case. 

Also, apart from that I also see that you include "v_0" in the last term of the cost function in Equation 13a. Isn't the last term "gamma*v_0*T_s" just a constant? So what is the objective of having that within the optimization problem, cause anyway constant don't affect the the value of optimization variables but just the global optima, given that problem is convex.

alexliniger, 2019-03-04:
Yes constant terms in the cost function have no influence. So if v_0 is constraint to a constant value we could also discard the term. 
However, v_0 is an input and only v_(-1) is constraint. The formulation gets quite complicated to wrap your head around. At the end, u_0 is actually in the second stage using the lifted state formulation.

PranjaLBiswas27, 2019-03-04:
Wow! That's pretty interesting. This is because of the lifted state formulation itself or MPCC formulation also plays a role in making u_0 available in the second stage? And if I understand correctly, second stage refers to the second prediction step. 

Do you have any specific reference to read for such lifted state formulations?

alexliniger, 2019-03-05:
I comes from the lifting. The idea is that we have the discrete time system x_k+1 = A x_k + B u_k and we want to constraint the difference in inputs (Δu) and add them to the cost we can do the following state augmentation
s_k = [x_k,u_k-1], v_k = Δu_k
and formulate the following linear system
s_k+1 = [A B;0 I] s_k + [B;I] v_k

Now u has a dynamic and is "driven" by Δu, if you write the system down you have
x_k+1 = A x_k + B (u_k-1 + Δu_k) 
u_k = u_k-1 + Δu_k
However, you see now that the state of the system is the last input and only in s_k+1 we have a u_k.

Thus if we constraint s_0 = \hat{s}_0 we do not constraint u_0 but u_-1. Further, u_0 = u_-1 + Δu_0 and s_1 = [x_1,u_0]

I hope this helps a bit
Best,
Alex

PranjaLBiswas27, 2019-03-05:
Thanks for the derivation. I think I do get an idea. But I have some queries as follows:

1) What exactly do you mean when you say "constraint" here, like in "if v_0 is constraint to a constant value" and " if we constraint s_0 = \hat{s}0". I assume it means the following,

> if v_0 is constraint to a constant value = if v_0 is a constant value

and

>  if we constraint s_0 = \hat{s}0 = putting constraint on the initial states of the lifted system and hence on the inputs of the system represented by Equations 12c and 13c

2) Just a small recap, please correct me if I am wrong. So, the prediction model (lifted system model) that is thus being considered within the optimization problem is the lifted system as in Equation 4 of the attached image. 
Because of that I understand that MPCC optimizes for Δu_k i.e. v_k instead of u_k directly. Thus, optimization of u_k comes out as a consequence of optimized Δu_k due to the relation between states and inputs in the prediction model. 
Now if I get it correctly, to impose input constraints on u_0 i.e. s_1, we select Δu_1 (the second prediction step i.e. k=2) instead of Δu_0, from the set of optimized lifted state inputs (Δu_0, ... , Δu_N) calculated by the MPC solver (generally the norm is of applying the first optimized input).

[image]

Is the explanation here at all correct? :P

alexliniger, 2019-03-05:
1. in MPC problems you always fix x_0 to the measured state, this is what I mean with s_0 = \hat{s}(0) -> \hat{s}(0) is the measured state and the last applied input.
2. Looks correct. 
Just to clarify something, the used solver optimizes all s_0,.....,s_N and v_0,.....,v_N-1. They are just linked by constraints, so you can have constraints on s_N and you don't need to consider all past inputs.

PranjaLBiswas27, 2019-03-05:
1) Thanks for pointing out the clarification. 

2) But just to put it differently, am I wrong in saying that MPC solver optimizes for the inputs (v_k) explicitly and as a consequence the states (s_k:[x_k,u_k]) get optimized implicitly.

3) However going back to the initial question of v_0 (virtual velocity along the path and not Δu_0) being constant or variable itself. From the above discussion we can say that _v_0_ (will refer to virtual velocity using italicized letters to avoid confusion from here on) is a variable. But what is the physical significance of including that in the cost function?

4) Also, since you say k=1,...,N for the cost function. In such a scenario in order to add any other costs for other inputs (steering and duty cycle), similar costs needs to be introduced?

alexliniger, 2019-03-05:
2. If the solver is converged yes. If not it depends on how the QP solver is implemented. If you are interested in optimization and optimization algorithms I suggest the Book: 'Convex Optimization' by Stephen Boyd 

3. _v_0_ is an optimization variable and the QP solver has to determine its value. Therefore, the cost does change when _v_0_ changes.

4. Yes, this is true but it is no problem and in the implementation here it is actually done. However, in a paper, there is often not a lot of space so you try to compress your notation. 

Maybe a general comment about this question, including constant cost (e.g., x_0' Q x_0) does not change the optimal solution of the problem so adding them or removing them does not change anything in practice.

PranjaLBiswas27, 2019-03-05:
Well, thanks a lot. I really appreciate you taking out the time to reply to the queries.

PranjaLBiswas27, 2019-04-09:
Hi @alexliniger,

I just wanted to clarify that are "q_c" and "q_l" scalars in Equation (13a)?

Kind Regards
Pranjal Biswas

alexliniger, 2019-04-09:
yes they are scalers, the errors are also scalar, the cost could also be written as q_c*e_{c,k}^2 + q_l*e_{l,k}^2

PranjaLBiswas27, 2019-07-19:
Hi @alexliniger ,

I was thinking over this again and I have the following questions:

1) We discussed that _v_0_ is a variable and needs to be optimized, but what I don't understand is that for k=1 (k varyies from 1 to N) _v_0_ is a state in itself due to the lifted state formulation. Hence _v_0_ is always initialized to some value at every time step while formulating the optimization problem, how then can _v_0_ act as a variable to be optimized. Also shown in the image below.

[image]

2) Since v_k  is the input of the lifted system it is the variable for the optimization problem. Now,
**k**  -----------------  **v_k i.e. _v_k_- _v_k-1_** (considering only 3rd input)
1  ----------------- _v_1_ - _v_0_
2  ----------------- _v_2_ - _v_1_
.
.
.
thus _v_k_ also becomes a variable implicitly given _v_k-1_ as a constant from state initialization for k=1 and from state dynamics for k=2, ..., N. Hence again how is _v_0_ variable. Also, I don't see a way to formulate it my optimization problem until its a state.

Please do let me know your views.

PranjaLBiswas27, 2019-07-19:
Hi @alexliniger,

I was working on implementation of rate based MPCC as discussed here. I tried two different approaches.

1) I applied the the predicted states ([x_1; u_0]) for k=0 (i.e. s_1) as input to my system. This gives me the correct results as we discussed.

2) I also tried to apply integration of v0 (receding horizon input) to my system, as v0 is just derivative of the actual system inputs. however this does not work and I am not able to understand why. Can you help me with whats wrong in this approach?

Looking forward to hear from you.

closed 2019-08-20 by PranjaLBiswas27