alexliniger/MPCC #10: Static and Dynamic Obstacles

PranjaLBiswas27, 2019-03-06:
Hi @alexliniger 

In this video (https://www.youtube.com/watch?v=JoHfJ6LEKVo), do you use the same approach as in this repository. Are the dynamic obstacles considered as freezed in time every sampling instant?

Kind Regards
Pranjal Biswas

alexliniger, 2019-03-06:
Yes, this video also shows the morphed constraints: https://www.youtube.com/watch?v=TMEz-Lfxc-E

You can see in the code that you could also give a prediction of the opponent car but assuming that the other car is a static obstacle is the simplest.

closed 2019-03-13 by alexliniger

PranjaLBiswas27, 2019-03-14:
Thanks, but where exactly can I find the obstacle avoidance part in the code?

Also, how exactly are predictions incorporated, some sort of lateral constraint over the prediction horizon, potential field for each obstacle over the prediction horizon?

alexliniger, 2019-03-18:
The predictions are set to constant here
https://github.com/alexliniger/MPCC/blob/6a5529b8453432e62d9806e2fd5bc020cf872c8a/Matlab/optimizer.m#L22
and the new borders are computed in 
https://github.com/alexliniger/MPCC/blob/master/Matlab/getNewBorders.m

The obstacle avoidance is done by modifying the track constraints.