alexliniger/MPCC #13: Unstable Contouring Error

PranjaLBiswas27, 2019-06-25:
Hi @alexliniger ,

I have been trying to implement MPCC for a while now, but I have an issue with the contouring error that it is exploding as shown in the Figure. Here it can be seen that lateral position of the real vehicle (red) and the predicted position (yellow) is deviating from the reference path at that arc-length (green).

[image]

I know it can be highly dependent on the way I have implemented things, but can you give some points from experience on where I might be going wrong which is making the error dynamics unstable. I am just using MATLAB NMPC toolbox to optimize for acceleration and steering rate without any input rate costs and constraints at the moment. Also, I am sure that the MPC vehicle dynamic model is correct from first principles and also it follows the high fidelity real world system response as can be seen in the plot. Also, the error cost is as simple as:

`J = sum(qc.*(E_C.^2)) + sum(ql.*(E_L.^2)) - sum((rv.*V_P).*Ts); %sum over full horizon`

so I doubt if that is wrong as well. Do you recommend looking at any other critical points which may be causing the issue? Looking forward to hear from you.

Kind Regards

Pranjal

alexliniger, 2019-06-27:
Only based on these plots it is hard to say what the issue is. Also, I don't know if these are predictions or the closed loop behavior. 
What I can say:
- I would include a cost on the inputs in the cost function, this is normally a good idea. 
- One thing I saw in your plots is that the heading is strange, at time zero the yellow and red curve are zero but not the green curve.

PranjaLBiswas27, 2019-06-27:
The plot actually shows the closed loop behaviour. Also w.r.t. the point you mentioned:
1) Can incorporating rate costs and constraints stabilize the errors?
2) It happens because starting of my reference path is not aligned with the initial position of the vehicle. So the reference path has an initial heading while I keep  the vehicle at 0 rad itself in order to see how it converges to the reference path.

alexliniger, 2019-06-28:
To see if the optimization problem is doing something reasonable you should also look at the predictions. For example, if the prediction is looking good and the closed loop is not performing well it indicates that something is wrong.

1. You should always add a cost on the inputs and/or the input rates. I don't know if it helps for your issues but it helps in general. You seem to be applying very aggressive inputs for example which is never a good sign

2. Are you sure that you did consider this change correctly when you are computing the lag and contouring error?

3. Maybe try to use the same model for simulation and prediction, it often helps to get it working with the simpler model and then use a more accurate model for the simulation.

PranjaLBiswas27, 2019-06-28:
When I see the predictions it indeed tries to converge to the reference parametrized path, but yes the vehicle does not behave similarly. From what I have observed, the MPCC implementation I have as of now acts more like a a PID controller which acts only after error starts accumulating from a visual perspective. So I guess there is an issue with closed loop performance but what I dont know as of now.

1.  Yes, my next step was to incorporate rates costs and constraints as I thought it should be possible to implement without one irrespective of how good the performance is. Have said that I also expected it to be stable which is not the case. So yup I will try with rates anyway now.

2. Do i explicitly need to consider offset of vehicle from the reference path during initialization? Should it converge to the path itself? I did not consider anything explicitly for that though.

3. Ok, I guess that can help me, thanks for suggesting that.

4. Yes I have state and input constraints to have realistic values, therefore inputs get saturated.

PranjaLBiswas27, 2019-06-30:
Hi @alexliniger 

My implementation indeed worked with the same model for simulation. After investigating more it indeed was an issue related to wrong state initialization of the high fidelity model w.r.t. the simple model used in MPCC. As of now, the vehicle is able to converge to the path from an initial offset. Thanks a lot for the mentioning critical points to consider.

Kind Regards

Pranjal Biswas

closed 2019-06-30 by PranjaLBiswas27