alexliniger/MPCC #16: Robotic implementation

hz3014, 2019-07-25:
Hi,

I have carefully followed your work and your cooperation with AMZ Driverless team. I managed to implement the Udacity MPC algorithm on my robot and i found conversion of MPCC from matlab to C++ very difficult especially the theta function. 

I was wondering how did you manage to implement the MPCC algorithm on RC car and driverless car without having the C++ version code.

alexliniger, 2019-07-25:
I assume with theta function you mean the spline. For the RC cars we used matlab to generate the spline and exported the parameters, and in C++ only had to perform the evaluation.

Generally, I do have a C and a C++ version of the controller and AMZ Driverless has their own implementation which is based on ForcesPro. Until now I just did not open source the C++ version, however, this is something a constantly evaluate.

MTDzi, 2019-07-25:
I, for one, would *really* appreciate a C++ version ;)

mywisdomfly, 2019-07-25:
I am also looking forward to the c++ version.

closed 2019-07-31 by hz3014