alexliniger/MPCC #19: Track Soft Constraint

PranjaLBiswas27, 2019-09-17:
Hi @alexliniger ,

I was trying to work with slack variables to soften the track constraint. I was referring to the following book regarding slack variables (https://web.stanford.edu/~boyd/cvxbook/bv_cvxbook.pdf, page 131, Eq 4.7). It says that to incorporate slack variables a new optimization variable is introduced, and the inequality constraint is transformed to an equality constraint. Although if I understand this correctly, will this not lead to the ego being always on the edge of the track? Because irrespective of how high the weight on slack variable be in the cost function, it will always try to satisfy the equality constraint (Eq 4.7).  Is there anything which I am missing in my understanding? Looking forward to hear from you.

Kind Regards

Pranjal

alexliniger, 2019-09-18:
I currently can only give a short answer, I think you mixed up transforming a qp into normal form, with soft constraints. In the normal form you want to eliminate all inequality constraints and just have equality constraints, this you achieve using slack variables.
Soft constraints one the other hand, does not eliminate the inequality constraints, it just reformulats the constraint with slack variables. Such that the problem has always a solution even if the problem would be infeasible.
What is done is that if the constraint is violated the slack becomes larger than zero and this again results in a cost. If the cost on the slack on is large enough, then the this formulation is the exact as the the optimizer can not improve the cost by violating the constraint. 
In practice however it is hard to find the correct value for the slack cost, as a too high cost results in a bad behavior when the constraint is violated.
Best, Alex

PranjaLBiswas27, 2019-09-18:
Thanks for the quick reply. Ok, that is one thing I need to look into. Although if new slack variables (s) are introduced to soften the constraints, I am not very clear of whether to treat them as states or inputs in the formulation. Because in my knowledge cost can be formulated only by states or inputs or their respective augmentations. 

Like in the paper referenced in this repository, the slack variable (s_k) used to soften the track constraint in Eq 14.a and Eq. 14.e, is not considered as state or input If I am correct. But then the cost includes the slack variables, how is that formulation possible. If you can put some light on that it will be great.

alexliniger, 2019-09-18:
at the end an MPC problem is an optimization problem, so states and inputs is a restriction we make when we talk about MPC but it is not necessary, you can just add optimization variables.

Second, if you use hpipm there is a special interface for soft constraints, it is not very well documented, but it is the only way to use soft constraints with hpipm.

PranjaLBiswas27, 2019-09-24:
> at the end an MPC problem is an optimization problem, so states and inputs is a restriction we make when we talk about MPC but it is not necessary, you can just add optimization variables

I don't know if I understand this fully, cause if I consider slack variable as a state I need to augment my state matrix somehow. This formulation of state matrix is not very clear in my mind. Or the other way can be I consider slack variable as an input which MPC optimizes and gives to me. So, i wanted your opinion in what is exactly the correct way to include slack (soft constraint). Also, I could not find any proper reference for soft constraint, so it will be great if you could direct me to any if available.

> Second, if you use hpipm there is a special interface for soft constraints, it is not very well documented, but it is the only way to use soft constraints with hpipm.

I am using Forces Pro, and as far as I have read there document I don't see any special interface for soft constraints.

alexliniger, 2019-09-30:
This is a Forces issue, however, if you use the discrete time interface, and not the continuous time approach, you can just add the slack as a "control" input. If you use the continuous time interface the ordering of states, control and slacks is more important as you have no control over which variable is used. If i remember correctly the ordering is [inputs, slacks, states].
If you want I can see if I can find some Forces NLP code, however i did not use it for quite some time since i do not have a license anymore.

chentairan, 2019-09-30:
Thank you for your patient answer, I have the same problem, and I don't know how to introduce the reference path in forces pro to build the objective function.

alexliniger, 2019-10-02:
@chentairan there are three ways to incorporate the reference path into ForcesNLP:
- you store the whole reference path (spline) in the cost function, if you have a long path you can also only give a part of the path and updating the parameters using the forces parameter framework 
- you approximate the contouring cost quadratically around the previous solution, outside of forces and just give forces the parameters of the approximation, for this you can use the code in this repo
- you approximate the path locally for each time step as a polynomial and give the parameters to forces
I know that all 3 methods work, for most experiments I used the second as it was the easiest given the code I had. However I think the first is the most elegant.

chentairan, 2019-10-02:
Hi @alexliniger 

Thank you for your patient reply. I tried the first method, but when I used Matlab's 'ppval' or 'interpolant' function to get the fitted coordinates in the cost function, the Forces Pro reports the error that does not apply to the Casadi object. I want to know how to get the position on the curve in the cost function without errors.

Kind Regards

Tairan

alexliniger, 2019-10-02:
Did you try writing your own ppval function
If you write the function in another file you have to import casadi if i remember correctly. This is most probably the reason the matlab ppval did not work

chentairan, 2019-10-02:
Thanks for the quick reply.
I didn't try writing my own ppval function.

I import casadi, but it still has an error:
`Error using histc
First Input must be a real non-sparse numeric array.`
Maybe it is because Matlab's data type is double but CasADI's is SX, MX.

PranjaLBiswas27, 2019-10-03:
> This is a Forces issue, however, if you use the discrete time interface, and not the continuous time approach, you can just add the slack as a "control" input. If you use the continuous time [...]
If you want I can see if I can find some Forces NLP code, however i did not use it for quite some time since i do not have a license anymore.

@alexliniger, ok  I will try to formulate the slack variable for the soft constraint as input and try to solve MPCC. Thanks a lot.

@chentairan,
If you want to use "ppval", according to me you should evaluate the spline beforehand and just pass the parameters of the spline to the cost function in FORCES Pro (similar to point 3 mentioned by @alexliniger ) or use other approaches as @alexliniger mentioned.  
Also, please try to create a new issue from next time cause I do not see how "Incorporating reference path in Cost Function" is same as "Track soft Constraint".

closed 2019-10-03 by PranjaLBiswas27