alexliniger/MPCC #23: Parameters of the car model

MarkSoviet, 2019-10-16:
I bought a totally same car as you described in your paper, but I got into trouble when identifying the parameters of the car model: 
1. The weight of the car is 186.5 g, which is much smaller (41 g) in your code, but the length and width is the same. I wonder if the parameters in your code is just an example case or true value you measured?
2. I read the paper "Analysis and control of high sideslip manoeuvres", which denoted the approach of identifying the Pacejka Tire Model. However this approach needs too much area, which cannot be realized on the track. Could you please share the details of the way for identifying the parameters? 

Waiting for your reply. Thank you.

alexliniger, 2019-10-16:
1. This is very strange, the parameter in the code are exactly the once I used for the real controller
2. I used a large area with the same material as the track for system identification. 
Here you can find my master thesis, explaining in more detail the system identification process:
https://docs.google.com/viewer?a=v&pid=sites&srcid=ZGVmYXVsdGRvbWFpbnxvcmNhcmFjZXJ8Z3g6MjY3YTJhYThiZmNkZjg5
For your case, have a look at my PhD thesis (https://www.research-collection.ethz.ch/handle/20.500.11850/302942), where I explain an alternative approach to identify the parameters, form the car driving around the track. However, you need a running controller for this.
Best,
Alex

alexliniger, 2019-10-16:
I would suggest using the mini-z, for some reasons:
- behaves better, the driving is less random
- larger, you can add some electronics, such as RF module, H-bridge, and so on, without designing your own pcb, just a arduino nano and some standard parts.
- the dnanos are no longer produced, so even if you find a car there are no spare parts anymore
- if you have an RCP track they also fit

closed 2019-10-20 by MarkSoviet