alexliniger/MPCC #24: Open loop or closed loop？

MarkSoviet, 2019-10-31:
Hi
I followed your work carefully and implemented it by using matlab-ros interface. But i failed to control the car well even for going straight forward. I wonder if your algorithm of the simulation is the same as the algorithm of your real car. Have you add some closed loop control method such as PID or just using the open loop MPC? 

Waiting for your reply. Thank you!

chentairan, 2019-10-31:
The jitter of MPC may be caused by the failure to find the optimal solution due to the long solution time. Generally, the performance of MATLAB is not enough to meet the real-time requirements, so you can try to use C interface solvers such as hpipm, FORCES Pro or IPOPT.

MarkSoviet, 2019-11-01:
> The jitter of MPC may be caused by the failure to find the optimal solution due to the long solution time. Generally, the performance of MATLAB is not enough to meet the real-time requirements, so you can try to use C interface solvers such as hpipm, FORCES Pro or IPOPT.

Thank you! In fact, hpipm solved for less than 2ms one time, which is fast enough I think. If there's no closed loop controller such as PID, I cannot guarantee the system stable.

alexliniger, 2019-11-01:
I think hpipm is not the issue, the limit is the whole time matlab takes. 
I use the same algorithm for the real cars, but the whole algorithm is implemented in c/c++. I don't use a low level controller, but depending on the car the model needs to be re-identified.
Stability and recursiv feasibility are its own topic in mpc, but a low level controller is not required.
Best, 
Alex

MarkSoviet, 2019-11-01:
> I think hpipm is not the issue, the limit is the whole time matlab takes.
> I use the same algorithm for the real cars, but the whole algorithm is implemented in c/c++. I don't use a low level [...]

Thanks for your reply! Would you make your c++ code open source? I'm quite interested in it.

USTfgaoaa, 2019-11-01:
@alexliniger 
I'm also quite interested in your real-time implementation. It would be a good reference for me to replicate this project.

closed 2019-11-28 by MarkSoviet