alexliniger/MPCC #26: Parameters of the full size car model

chentairan, 2020-01-10:
Hi @alexliniger 

After I customize my track and bound, I try to modify the model but found that the exitflag of the solver was 2. I think it should be caused by the improper vehicle model parameters I set. Would you like to ask if you have full-size vehicle model parameters for simulation?

Waiting for your reply. Thank you !

Tairan Chen

alexliniger, 2020-01-13:
I noticed the problem when i tried the same. I assume normalizing the state and input should help, at least it did for the Matlab code.

Another solution could be transforming the origin of the global coordinate system to X_0, Y_0, in the MPC. The last approach is to use the warm start feature in hpipm. 

I try to look into them once I have time.

Best,
Alex

chentairan, 2020-01-13:
I found that if I didn't change the model parameters, I could get exitflag equal to 0.

Do you mean that I need to normalize both state and input to eliminate the influence of changes in parameters such as Cm1 and Iz in the model?

Best,
Tairan

alexliniger, 2020-01-13:
The issue is that a full scale car results in a bad scaled problem, you have positions in the range of 1000m and inputs and states that are only in a range of +/- 1. Thus it often makes sense to scale everything to +/-1 as I did in the Matlab code.
Since the biggest problem are the positions, moving the coordinate system or using an initial guess most probably also fixes the problem. And as usual everything together is what one should do. 
Best,
Alex

chentairan, 2020-01-13:
I see. Thank you for your reply !

Best,
Tairan

closed 2020-01-13 by chentairan

chentairan, 2020-05-19:
In the acado version I reproduced, I have not added tire constraints, and the solver seems to be able to converge.
[image]

alexliniger, 2020-05-19:
Thanks, this means there is a small bug in my code or using the affine system is causing issues (this is currently something I suspect as it may work good for the miniature cars where the states are very small and all in a similar scale but it may causes issues for the full sized cars).
Acodo is using the following approximation x_{k+1} = A_k x_k + B_k u_k