alexliniger/MPCC #29: Positive Definiteness of the Quadratic Part of Linearized Tracking Error

AmineElhafsi, 2020-01-22:
Hi,

I'm working on performing the Taylor expansion of the first two terms of equations 13a to get the first two terms of 14a. I'm using an automatic differentiation package to get the c_k and Gamma_k terms in equation 14a however I'm finding that the quadratic part is not necessarily positive semi-definite. Did you employ any specific techniques to ensure that the quadratic part was psd?

Also, just a quick question regarding notation, in the first two terms under the summation in equation 14a, should the vectors [x_k theta_A,k] be interpreted to represent [x_k-x_0,k theta_A,k-theta_A0, k], where the 0 subscripted terms represent the points about which we are Taylor expanding? Or am I misunderstanding?

Thanks!

alexliniger, 2020-01-24:
I am using Gauss newton hessian approximation method 
https://github.com/alexliniger/MPCC/blob/af5adf793fb01b2f844016b74b9efef87d8fb988/C%2B%2B/Cost/cost.cpp#L121
maybe the approach is simpler to understand if you look at the beta cost.
https://github.com/alexliniger/MPCC/blob/af5adf793fb01b2f844016b74b9efef87d8fb988/C%2B%2B/Cost/cost.cpp#L85

The idea is that for nonlinear least square costs, you don't have to compute the hessian but you can approximate it using the jacobian. There are two advantages, first it is computationally cheap, second it is guaranteed PSD. 

14a is correct, the x_0 and thata_0 terms are considered in the linear cost as well as in the constant term which can be neglected in an optimization problem. 

Hope that helps,
Alex

closed 2020-01-28 by AmineElhafsi

LoweDavince, 2021-01-15:
Hi @alexliniger ，
I don't understand For SQP approximation  why following line should add  the ` - d_contouring_error*stateToVector(x)` part?
`const double contouring_error_zero = error_info.error(0) - d_contouring_error*stateToVector(x);`

Thanks!

alexliniger, 2021-02-05:
y = f(x) ~ df/dx(x-x0) + f(x0) = df/dx x + [f(x0) - df/dx x0] 
The contouring error zero term is the term in the [ ] brackets