alexliniger/MPCC #30: The full size parameters of the c++

Serenos, 2020-02-07:
Hello, Alex

I have been trying to use the parameters of the full_size model in the Matlab implementation to run the C++ implementation, but the solver exit flag=2.

I did the normalization as you did in the Matlab implementation.

Have you try the same thing or can you give me some advice ?

Hope for replies, thanks.

Best, 

Lixiang

alexliniger, 2020-02-10:
The C++ code has some small differences, which changes the tuning. There is also a small error I did found but not yet push the fix, and I also did implemented the normalization. I want to test everything this week and push the new code. 
Generally, one issue is the tuning of the controller which can have a influence on the convergence of the solver, the other is the curvature which is currently fixed with a hack that is track dependent. I push this fix asap. 
Best,
Alex

chentairan, 2020-02-16:
Hi Alex,
I tried your latest code. Modify the `model.json` file with the parameters of **FSSIM** which I think is the same as the paper's you mentioned. But the result is still exitflag 2. It seems like I missed some details. Do you have any good parameter adjustment suggestions?

```
{
  "Cm1" : 5000,
  "Cm2" : 172,

  "Cr0" : 180,
  "Cr2" : 0.7,

  "Br" 	: 12.56,
  "Cr" 	: -1.38,
  "Dr" 	: 1.60,

  "Bf"	: 12.56,
  "Cf" 	: -1.38,
  "Df" 	: 1.60,

  "m" 	: 100,
  "Iz" 	: 110,
  "lf" 	: 1.22,
  "lr" 	: 1.22,

  "car_l": 2,
  "car_w": 1,

  "g"   : 9.81,

  "R_in" : 1,
  "R_out" : 1,

  "max_dist_proj": 1,

  "E_long" : 0.9,
  "E_eps" : 0.95,

  "maxAlpha" : 0.6,

  "initial_velocity" : 1.0,
  "s_trust_region" : 0.8,

  "vx_zero": 0.3
}
```

alexliniger, 2020-05-19:
There are some errors in the parameters you have here, since the AMZ is using a slight modification of the model I use, but even after fixing these errors the solver does not converge.
It is on my do to list trying to fix this issue, but it turned out to be harder than first anticipated.
Best,
Alex

chentairan, 2020-05-19:
it's true. I tried to modify the Df and Dr, and the solver did not converge, either.
Thanks,
Tairan

KarasevPavel, 2020-05-27:
Hi! I have solved the problem by using model parameters  from the FSSIM with tire coefficients Сf,Cr,Bf,Br from the initial model.json file and with Df = 4820 and Dr = 4820. I know, that these parameters are not quiet right, but You can use them as initial. Do not forget to change bounds in bounds.json file. Problem was in very small Ffy and Fry forces, so tire constraint did not work properly and car did not have an opportunity to corner with so small latteral forces.

kampountolas, 2020-07-13:
> Hi! I have solved the problem by using model parameters from the FSSIM with tire coefficients Сf,Cr,Bf,Br from the initial model.json file and with Df = 4820 and Dr = 4820. I know, that these [...]

Hi we have the same issue with a full scale car in MPCC (C++ version). I was wondering can you share with us a full set of working files for the full scale car?

Thanks, Kostas

KarasevPavel, 2020-07-15:
> Hi we have the same issue with a full scale car in MPCC (C++ version). I was wondering can you share with us a full set of working files for the full scale car?
>
> Thanks, Kostas

Hi! I have pushed a project to my local rep https://github.com/KarasevPavel/MPCC-FullScale, there is a small oval track there, as an example. Enjoy, please.

Regards,
Pavel

alexliniger, 2020-07-21:
I implemented a more standard SQP method which is numerically more stable, as it only computes search directions and not the full state. Now it is working, and on top of that the computational performance is even slightly better. 
You can find the new version in the branch fullsize, where I use a formula student typ car and a full sized race track I got from a user. 
Best,
Alex

closed 2020-07-21 by alexliniger

kampountolas, 2020-07-21:
Hi Alex, Thanks a lot for updating the code to the full-size car. We are currently trying a number of tracks with our formula specs.

Best, Kostas

kampountolas, 2020-07-21:
> Hi! I have pushed a project to my local rep https://github.com/KarasevPavel/MPCC-FullScale, there is a small oval track there, as an example. Enjoy, please.
>
> Regards,
> Pavel

Hi Pavel, thanks for your swift reply and for sharing your code. It seems works for a number of tracks and it fails for some other.

Best, Kostas