alexliniger/MPCC #31: A question about obstacle avoidance

LiJiangnanBit, 2020-02-08:
Hi, Alex
I have 2 questions:
1. In the track constraints of the MPCC problem:
[image] 
for any k, the track boundary (corridor) is computed according to the **s** of the corresponding initial guess point. But obviously, the solution might be different with the initial guess, so at the kth point of the solution, the **s** changes, thus the track boundary is actually not the value used in our constraints. 
So it seems that even when the track constraints are satisfied, the predicted trajectory may have collision with obstacles? How do you deal with this situation?
2. In obstacle avoidance, the vehicle is treated as a point without considering its heading. Does this matter when the corridor becomes narrow? Sometimes even if the point stays within the corridor, part of the vehicle might be out of the corridor because of its length and heading.
Thanks!

alexliniger, 2020-02-10:
1. Since the track constraints are approximated at each step in the SQP scheme, if the SQP converges, the track constraints are fulfilled. But if you stop early (as we do here) you could be right, it very much depends on how good a linear approximation works for your actual constraints. 

2. To deal with this at the moment I use a safety margin. The issue is that the safety margin makes the problem conservative. So there is a certain trade-off. It would be interesting to implement orientation dependent constraints, but I did not yet found the time to do it.

LiJiangnanBit, 2020-02-10:
Thanks for your answer :smile:

closed 2020-02-10 by LiJiangnanBit