alexliniger/MPCC #32: Discretized Model Scalling

pulsipher, 2020-02-17:
I follow how the model dynamics are linearized/discretized to form `Ad`, `Bd`, and `gd`. However, could you provide some more detail about the following scaling step that seems to lift the problem:
```matlab
% constructing augmented system with state-input scaling
    Ak = [MPC_vars.Tx*Ad*MPC_vars.invTx MPC_vars.Tx*Bd*MPC_vars.invTu; zeros(nu,nx) eye(nu)];
    Bk = [MPC_vars.Tx*Bd*MPC_vars.invTu;eye(nu)];
    gk = [MPC_vars.Tx*gd;zeros(nu,1)];
```
I don't quite follow why we increase the dimension of x by 3.

alexliniger, 2020-02-17:
https://github.com/alexliniger/MPCC/blob/949909291a0c32c3043017cc4f9caab37f01b0ce/Matlab/getMPCmatrices.m#L35
The idea is that the MPC does not compute the inputs but how much to change the inputs, this allows for rate constraints and rate costs, while still maintaining the structure necessary for hpipm.

pulsipher, 2020-02-17:
Thanks, I missed that note.

closed 2020-02-17 by pulsipher