alexliniger/MPCC #33: Linearization adjustment

pulsipher, 2020-03-03:
Could you please explain the purpose of the adjustment present in https://github.com/alexliniger/MPCC/blob/master/Matlab/DiscretizedLinearizedModel.m#L54-L62. Thanks

alexliniger, 2020-03-04:
The issue is that the model has a singularity at v_x = 0, which is clear when you look at the slip angle definition. This results in the errors even if the car is driving very slow. 
For the miniature race car application, this is not a bit issue since the cars are nearly never driving slow. To still avoid issues when starting, I added this statement. 
The if loop does two things:
- if the car is driving slower than 0.5 m/s, v_y, r, and delta are set zero for the linearization, which corresponds to assuming small steering angles and slip angles (linear tire forces).
- if the car is even driving slower than 0.3 m/s, additionally the forward velocity is set to 0.3.

This is a hack and if driving slow is important I suggest using a kinematic car model. Or the mixed kinematic-dynamic model proposed in the AMZ Driverless paper.

Best,
Alex

pulsipher, 2020-03-04:
Thank you, that clear things up.

closed 2020-03-04 by pulsipher