alexliniger/MPCC #35: Comparison results between different Solvers

NUDTUGVexplorer, 2020-03-22:
Hi, Alex
Do you have the comparison results of the solver performance between Hpipm and FORCES Pro?
Which is better?

Thanks.

alexliniger, 2020-04-01:
I never did an in-depth comparison. Generally, my observation is that Forces QP is slower than Hpipm, note this is the old once public FORCES QP solver, not Forces Pro. 
Force Pro NLP is a bit of an unfair comparison because we do just a fixed number of SQP steps, but solving the Forces NLP problem takes at most about 20-30 ms (with horizon 40, and a limited number of iterations), one SQP iteration with hpipm takes at most 3-4ms. Thus if you know how many SQP steps you need you can find out which is faster.
I would like to note that Forces Pro NLP also comes with a lot of advantages, it is more user friendly than writing your own SQP approach with hpipm and it normally works out of the box. Maybe I can suggest looking into acados or MATMPC if you look for an open-source alternative, both can use hpipm as a QP solver. 

Best,
Alex

chentairan, 2020-04-01:
I wonder if IPOPT will be slower than other solvers. Do you have any practical experience?

Thanks,
Tairan

NUDTUGVexplorer, 2020-04-02:
Got it. Thanks Alex.

alexliniger, 2020-04-02:
@chentairan IPOPT is normally not ideal for real-time MPC, it is just too slow (when I remember correctly about 10 times slower than FORCES NLP). However, it can be very helpful for design problems with long horizons or in general a lot of optimization variables. 
Also, real-time MPC solvers such as hpipm and forces require a special structure of the optimization problem, IPOPT does not have this restriction

chentairan, 2020-04-02:
Thanks for your reply. I will first try to use acados solver for MPC implementation. Maybe later I can help you add an acados qp interface👍🏻

Best,
Tairan

closed 2020-05-19 by alexliniger