alexliniger/MPCC #36: Extremely small change of delta

chentairan, 2020-03-28:
Hi @alexliniger ，

I modified the model parameters and boundary conditions, and kept the cost parameters. Try to test on a circular track, but after 600 iterations, the delta change is extremely small. Do you know the reason?

[image]

Thanks,
Tairan

alexliniger, 2020-04-01:
I currently try to find a way to run MPCC with the full-scale parameters of the Matlab implementation. But it turned out to be harder than expected. 

One way to somehow get it working is to always use the reset initial guess that starts from the center line with a fixed speed (change that speed to at least 5m/s), and using about 5+ SQP iterations with a SQP mixing of ~0.25. Also playing around with the cost function can help.

This is not how it should work, and I will try to fix this behavior once I have time, but it turned out to be harder to find than expected. 

Best,
Alex

alexliniger, 2020-07-21:
Can you try this again with the new fullsize branch.
Best,
Alex

chentairan, 2020-07-21:
That's worked, thanks !

closed 2020-07-21 by chentairan