alexliniger/MPCC #37: About the local coordinate system conversion

chentairan, 2020-05-15:
Hi @alexliniger 

I am using acado to reproduce the MPCC, it seems that it can simply run in the global coordinate system, but when `phi = M_PI` produces weird problems like the figure below.

[image]

So I plan to convert the coordinate system of the mpc problem to the vehicle coordinate system, but it seems that the `updateInitialGuess` step needs a lot of coordinate system conversion. I have not tested whether it is feasible, I would like to ask if you have any good ways to reduce the transformation of a large number of coordinate systems when converted into a vehicle coordinate system?

Best,
Tairan

alexliniger, 2020-05-19:
I don't really know what you mean with a vehicle coordinate system? A frenet frame or curvilinear coordinates? 
The issue you show looks like you have a jump in the angle during your horizon. In the updated initial guess, I check that the initial guess never has jumps, it is allowed to leave the interval [0, 2 pi] but it is not allowed to have jumps. 
If you need to perform a lot of coordinate changes it can be feasible if you have a good initial guess of where you are, the newton method I use to project onto a spline should be very quick in finding such a projection, so you can do quite a lot of them. 

Best,
Alex

chentairan, 2020-05-19:
Vehicle coordinate system means transforming the origin of the global coordinate system to X_0, Y_0, in the MPC.
I think current `updateInitialGuess` function is to shift the step of the states `x` and control `u`, but if using a local frame, simple shift seems unreasonable. Maybe a lot of rotation and translation is needed ?

Best,
Tairan

alexliniger, 2020-05-22:
Yes this can be quite exhaustive, especially since for some of the cost terms you need the global position.
Since you implemented it in ACADO I assume it should not make b big difference having a local or global frame. But if you are only worried about the computational cost I assume this should not be an issue.

chentairan, 2020-05-22:
Thanks, I will have a try later.

closed 2020-05-22 by chentairan