alexliniger/MPCC #38: Work of plotBox method of Plotting class

KarasevPavel, 2020-05-28:
Hello, Alex!

Yours  <b>plotBox</b>  method in class <b>Plotting</b> does not work properly, because you need to divide body_xl, body_xw, body_yl and body_yw by two before appending and subtracting them from X and Y coordinates of the car.

Regards,
Pavel.

alexliniger, 2020-05-28:
I just define car_l and car_w as half the car length and width .

https://github.com/alexliniger/MPCC/blob/b2a2999d5541934b4eec4b30103c05681172c998/C%2B%2B/Params/model.json#L22

I went with this definition because nearly all the time you need the size of the car, you only need half the width and half the length. 
Unfortunately it is not possible to add comments to the json file. Otherwise i would have added a comment.

KarasevPavel, 2020-05-28:
Okay, Thank you!

Regards, 
Pavel.

closed 2020-05-28 by KarasevPavel