alexliniger/MPCC #39: ./install.sh failed

DreamonZhu, 2020-07-28:
when I tried to run ./install.sh to install the project of c++ version, an error occured. 
The error info was mainly about "CMake error in hpipm/test_problems/CmakeLists.txt". In fact, there was a target "hpipm" containing relative path in its INTERFACE_INCLUDE_DIRECTORIES: '../../BLASFEO/lib/include'. I guess maybe it was because of the version of cmake, I couldn't use the relative path in CmakeList.txt. And I was just not able to reverse the relevant code. Anyone could help me? Thank you so much.

chentairan, 2020-07-28:
Hi Zhu,

Maybe I can help you. I meet the same problems.

First you can set `-DBLASFEO_PATH` to an absolute path, for me I set `-DBLASFEO_PATH=/home/chentairan/Desktop/MPCC/C++/External/blasfeo/lib`.

Second the `matplotlib` package also has some problems when building. You can checkout to `61da78f` commit, and try again.

Best,
Tairan Chen

alexliniger, 2020-07-28:
This was also my solution, to fix this issue.
The issue with matplotlib I had was related to python 2.7 and Ubuntu 20.04. Which required installing pip2 from scratch.

Best,
Alex

DreamonZhu, 2020-07-29:
> Hi Zhu,
>
> Maybe I can help you. I meet the same problems.
>
> First you can set `-DBLASFEO_PATH` to an absolute path, for me I set [...]

> Hi, Tairan
> First, thank you so much.
> And I want to make sure what you mean is that I need to add an global varible into ~/.bashrc, then I just use this varible in my Cmakefile.txt, right?

chentairan, 2020-07-29:
No need to modify the global variable, you just need to modify this line https://github.com/alexliniger/MPCC/blob/master/C%2B%2B/install.sh#L53

DreamonZhu, 2020-07-29:
> No need to modify the global variable, you just need to modify this line https://github.com/alexliniger/MPCC/blob/master/C%2B%2B/install.sh#L53

> Ok, I've successfully installed it and run it. Thank you.

closed 2020-07-29 by DreamonZhu