alexliniger/MPCC #46: normalized state and input

DreamonZhu, 2020-08-31:
Hi alexliniger,
https://github.com/alexliniger/MPCC/blob/3ec7805e7a176c57250c1a414a6d2b871e472d74/C%2B%2B/MPC/mpc.cpp#L73
In this place, you did the normalization for the problem. And I guess it was done by replacig "x'=T_x \times x", "u'=T_u \times u" and "s'=T_s*s".
Then,
https://github.com/alexliniger/MPCC/blob/3ec7805e7a176c57250c1a414a6d2b871e472d74/C%2B%2B/MPC/mpc.cpp#L235
but here you just dealt with state x while doing nothing with input x and slack variable s? I don't know why?

alexliniger, 2020-08-31:
The first part is normalizing the matrices of the optimization problem and thereby indirectly the optimization variables, the second line is normalizing the "measured" state x0
This function
https://github.com/alexliniger/MPCC/blob/3ec7805e7a176c57250c1a414a6d2b871e472d74/C%2B%2B/MPC/mpc.cpp#L119
is then de-normalizing all the variables back to their metric values.

Normalizing the optimization variables is only depending on the matrices and the initial state x0, so I don't have to normalize anything else. But once I get the solution I have to get it back to my normal space.

I hope this helps, otherwise I suggest you just write down all the math for normalizing a QP.

DreamonZhu, 2020-09-01:
Thanks a lot. I got it.

closed 2020-09-01 by DreamonZhu