alexliniger/MPCC #47: sqp_related

DreamonZhu, 2020-08-31:
Hi alexliniger,
https://github.com/alexliniger/MPCC/blob/3ec7805e7a176c57250c1a414a6d2b871e472d74/C%2B%2B/MPC/mpc.cpp#L238#L239
could you please tell me what n_no_solves_sqp_ means?
https://github.com/alexliniger/MPCC/blob/3ec7805e7a176c57250c1a414a6d2b871e472d74/C%2B%2B/MPC/mpc.cpp#L244-#L251
Also I am so sorry I could ot understand the logic of the code here? Hope you could explain it to me? I believe it also has something to do with the meaning of n_no_solves_sqp_ . But I just am not able to get it.
Thank you so much first.

alexliniger, 2020-08-31:
The question is what to do if your QP does not converge, n_no_solves_sqp_ is just the number of times the solver did not converge during one SQP solve. Depending on how often this happens and how often a similar problem occurred in the time steps before a default initial guess is used and not the previous solution. 
This is not a very good solution, and one should do something better there but if everything works well it is not really needed.
Best,
Alex

DreamonZhu, 2020-09-01:
Did you have any reference books or documents which introduce the usage or API of hpipm or some suggestions to learn hpipm ? I just want to know hpipm a little more. Anyway, thanks very much.

alexliniger, 2020-09-03:
https://github.com/giaf/hpipm there you can find everything (it is not a lot)

closed 2020-09-29 by alexliniger