alexliniger/MPCC #52: Normalisation for QP

dwirya, 2020-10-19:
Hi Alex,

I'm currently new to optimisation theory and I don't quite follow why we need to normalise the QP problem as seen in your C++ code. Does this result in obtaining the solution faster ? Additionally, how did you get the normalisation parameters ? Thank you in advance.

Regards,
Dennis

alexliniger, 2020-10-22:
Sometimes normalization can help, especially if some of the states and actions are in completely different ranges, but this is more related to numerical issues.
The normalization parameters are just chosen such that the states are in the +/- 1 range. But generally it is not really necessary, in the full size version it works good without normalization. The main branch does not work well for full sized cars.
Best,
Alex

dwirya, 2020-10-25:
I see, thank you for the clarification!

Kind regards,
Dennis

closed 2020-10-25 by dwirya