alexliniger/MPCC #53: Waypoints-  input explanation

Akhil2194, 2020-11-20:
The way points in the C++/Params/track.json already consists of the racing line (X and Y co-ordinate ) as input which the RC cars follows. Shouldn't the  trajectory (X and Y co-ordinates ) be calculated by the cost functions as an output ?

alexliniger, 2020-11-22:
No the track.json file for the miniature race cars is the middle line. MPCC is a path following controller, thus, it needs a reference path it should roughly follow, we often use the middle line, but an ideal line does also work. For the full-size car implementation, I did only have a rough ideal line, but still if you allow MPCC to deviate it will cut the corners more than the ideal line. Finally, there is a difference between an ideal path and the corresponding full state trajectory.

P.S. using the ideal line actually helps the performance as we showed in a IROS 2020 paper: https://arxiv.org/abs/2003.04882

closed 2020-11-30 by alexliniger

Akhil2194, 2021-05-07:
@alexliniger  hi. i was trying to test the code with separate waypoints without the inner and outer boundaries . It worked for one set of waypoints bu not for the other. I am getting an error - 

bounds initialized
MPCC: /home/akhilsharma/MPCC/C++/External/Eigen/Eigen/src/Core/DenseCoeffsBase.h:181: Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index) const [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType = const double&; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.
Aborted (core dumped)

The waypoint file is attached here.

[pretty.txt](https://github.com/alexliniger/MPCC/files/6439288/pretty.txt)

alexliniger, 2021-05-07:
Your file has 1178 element in X but only 1169 in Y
The length of X, and Y has to be the same