alexliniger/MPCC #54: New waypoints as input

Akhil2194, 2020-12-08:
I am trying to use the full size branch for my application. I have imported waypoints externally. But i am getting an message after the computation has been done and the trajectory generated is does not adheres to the curvature present in the waypoints  :
this is message 
ipm arg size = 0
comp time = 0.001643
exitflag 2
ipm iter = 8
dist too large
Could you please guide me and tell me other changes needed for a full application on full size car simulation.

alexliniger, 2020-12-09:
exitflag 2 means the QP solver did not converge, and 'dist too large' means your starting position is quite far away from the reference path. Maybe you can reduce the number of simulation steps to 1 and see how the path and the current position looks like.
Without your reference path (waypoints) I can not tell you what the issue is. Also does this happen at the first step or further along the path? Do you have a high curvature path?

Akhil2194, 2020-12-09:
Thanks for the quick response . **exitflag 2** occurs throughout the route and the '**dist too large**' starts as soon as the road curvature starts and then continues. And No , I don't have any high curvature path. I am attaching a part of the route for your reference. Please let me know the areas to modify/. 
[dummy track.zip](https://github.com/alexliniger/MPCC/files/5665240/dummy.track.zip)

alexliniger, 2020-12-09:
I assume that track is a closed race track, so this may cause some of the issues

alexliniger, 2020-12-09:
The issue is that the bounds on X-Y are +/- 3000 m and your track is outside of these bounds, changing the bounds and it works. Still be careful about the unwrapping in the progress state.

Akhil2194, 2020-12-10:
Its working.Thanks for your quick help.  Is it possible to breakdown the state vector so as to update the state at every instance from a simulator.?

alexliniger, 2020-12-10:
Yes, this line 
https://github.com/alexliniger/MPCC/blob/84cc61d628a165a424c805bbe071fe96b88da2d0/C%2B%2B/main.cpp#L69
is a simulator, so you can replace this simple RK4 integrator with your simulator. The best way is to write a ros node which is the MPCC and that communicates with the simulator.

Akhil2194, 2020-12-14:
Thank you.
Sorry for consecutive queries. 
Is the whole path necessary at the very initial starting for optimized trajectory or can a real time trajectory which a vehicle sensors can detect (for suppose next 100  time step) enough from the simulator ? is there an alternative ?

alexliniger, 2020-12-16:
It is possible, you need to solve two things, first you need the "splinify" the trajectory you would like to follow, this is mainly a bit of coding effort and there are better tools to fit splines into curves than the one I implemented here. The second challenge is that you need to transfer the solution form the previous solve to your new path, since you use it as an initial guess. Again this is coding effort as it needs changes to my initial guess implementation, and if it is done wrong it can cause convergence problems.

closed 2021-01-15 by Akhil2194