alexliniger/MPCC #58: Repo for work with GP

seanjkanderson, 2021-03-31:
Hi Alex,

I'm working on an implementation of the theory in `Cautious NMPC with Gaussian process dynamics for autonomous miniature race cars` and have been having a hard time getting it to work in FORCES PRO. Is there a repo for this? I shot you a couple emails as well but I thought I'd try here as well.

Thanks!
Sean

alexliniger, 2021-03-31:
I responded your email thanks for the reminder.

closed 2021-03-31 by alexliniger

ShaoshuSu, 2022-07-08:
Hi Alex, I am also interested in your GP project which is related to your paper `Learning-based model predictive control for autonomous racing` and `Cautious NMPC with Gaussian process dynamics for autonomous miniature race cars`, may I ask for the availability of this repo as well? Thanks!