alexliniger/MPCC #59: Steering constraint to avoid crash

AIDRI, 2021-04-07:
Hello,
I am actually checking the project, but i was thinking about the steering constraints. Sometimes, I can see that the car make turn with a high steering angle (20°) and a high speed. I think it s impossible that a car can do that, so I have a little question : 
Can we manage the maximum steering angle using your method ?

alexliniger, 2021-04-07:
The small cars can have quite high side slip angles, in the case of the full scale car this should be avoided. 
To answer your question, yes the steering can be constrained to whatever range you like. In the C++ code it is in the bounds.json file and in the matlab version it should also be a config file. 
However, it is maybe better to penalize this behavior with the beta cost which is implemented in C++, this is exactly one of the reasons we added this cost.

AIDRI, 2021-04-07:
ok, thanks for your answer.
I am actually trying to make a script to run a car in the simulator, but the car have a mass of 2 tones, that s why I d like to have steering constraint. 
Btw, I checked the Matlab script, but i only see vars to change the cost of change of steering. If you have time, can you help me finding the var to do that ?

alexliniger, 2021-04-07:
https://github.com/alexliniger/MPCC/blob/84cc61d628a165a424c805bbe071fe96b88da2d0/Matlab/getMPC_vars.m#L36
https://github.com/alexliniger/MPCC/blob/84cc61d628a165a424c805bbe071fe96b88da2d0/Matlab/getMPC_vars.m#L39

you can change it in these two lines, 0.35 is the current steering limit. you need to change it in both lines.

However, i am not sure how good it works for a full sized car, in the C++ code i had to change quite a lot to make it work properly.

AIDRI, 2021-04-07:
Ok, i wasn't looking at the good script. 
Thanks for your answer (3 min reaction time haha), and good luck for the next improvements you would like to make 👍

closed 2021-04-07 by AIDRI