alexliniger/MPCC #71: some questions about tractive force at the rear force

dreamonJure, 2021-11-28:
Hi @alexliniger 

First Thank you for your excellent work and code. I got a lot from it. So appreciate it. 

I notice that it is an interesting way when computing the tractive force at the rear wheel F_r, x, which was defined as (C_m1 - C_m2 * v_x) * d. And I wondered how to get this simplistic equation. I've read two papers "Optimization‐based autonomous racing of 1:43 scale RC cars" and "AMZ Driverless: The Full Autonomous Racing System". Unfortunately, I didn't find the reference to it.

Could you give me instructions?

Best Regards,
Alex

alexliniger, 2021-11-29:
It is a torque DC motor model using steady state simplifications.

The moment produced by a motor is `k_m*i`, the current can be approximated by `i = (V_bat - V_ind)/R`, where `V_bat` is the battery input and `V_ind` is the inductive voltage of the motor. This is only true if the duty cycle of the pwm signal is 1, if it is not zero, it is this value during the one phase and 0 otherwise.
Thus on average we have the torque of the motor to by `T = k_m*d*(V_bat - V_ind)/R`. The inductive current by `V_ind = k_m*omega`, the motor speed omega is again proportional to the wheel speed. Finally, the force is again proportional to the the torque, which results in `F = c*T = c_t*k_m*d*(V_bat - k_m*c_w*v_x)/R`, where `c_t` and `c_w` depend on the gearbox and the wheel radius. 

Finally this results in a the equation `F_x = (C_m1 - C_m2*v)*d`, where `C_m1` and `C_m2` are identified by driving data.

I hope this helps,
Alex

dreamonJure, 2021-11-29:
Thanks for giving the derivation. And So could I interpret ```d``` as the duty cycle of the pwm signal here?

alexliniger, 2021-11-29:
Yes exactly. 
From the README `The inputs are the duty cycle d to the drive train, the steering angle delta and the velocity along the reference path v_theta (which is an approximation of the velocities projected onto the path)`

dreamonJure, 2021-11-30:
OK. And in "AMZ Driverless: The Full Autonomous Racing System",  there also exists a ```D``` which is driving command replicating the pedals of a driver. Does it has the equivalent meaning as duty cycle ```d```? 
Furthermore, if I want to equip my own model car, how could I realize this physical actuator? Could you please give me more details and suggestions?

alexliniger, 2021-11-30:
The meaning of the driver command is very similar, but it is a bit more simplistic. The model neglects the induced voltage, so it becomes `F_x = C*D`. In other words the force is just proportional to the [-1,1] driver command. This simple model is possible because there is a low level controller which maps the driver command to the actual commands of the motor. 

With a model car you most probably have a somehow similar behavior, so at the end the challenge is mainly to find the constant `C`.

dreamonJure, 2021-12-01:
Thank you so much.

closed 2021-12-01 by dreamonJure