alexliniger/MPCC #72: C++ build error

Geonhee-LEE, 2022-03-06:
Hi, @alexliniger.
Thanks for your contribution.
I am interested in MPCC and want to run on C++ execution on the Ubuntu 18.04.  According to your README, I tried to make the CMakeLists but failed like following:

[image]

After failed to make, I changed the CMakeLists and finally succeeded to build. But, I encountered the problem when I launched with './MPCC'. I found that problem is associated with [matplotlib-cpp](https://github.com/lava/matplotlib-cpp). However, I couldn't find the solution although strive to solve this problem with some modification and trials.
[image]

Let me know what is the problem if possible.

alexliniger, 2022-03-06:
the two branches use different approaches for matplotlib-cpp in the CMake file, can you try out the other one as well. I will have a look as soon as possible. But over the years I had also several times issues with matplotlib-cpp.
Best,
Alex

Geonhee-LEE, 2022-03-06:
Thanks for your reply.
I tried to change from `master` branch to `/wip/install-target` branch. It also happened a same problem.

Geonhee-LEE, 2022-03-06:
The python configurations:
[image]

[image]

Geonhee-LEE, 2022-03-07:
Hi, @alexliniger 
I finally correct the problem for matplotlib.
But, I want to launch the MPCC path tracking online like README :

https://github.com/alexliniger/MPCC/blob/master/Images/MPC_sim.gif

closed 2022-03-07 by Geonhee-LEE