alexliniger/MPCC #82: Build instructions for new CppAD-codegen

ckurtz22, 2023-04-25:
There is a lack of build instructions or process for the new CppAD-codegen branch (or docs in general for that matter :P ). 
I've managed to fumble most of the way so far on my own to get a "successful" build but now getting runtime errors when I run the main executable MPCC. Also, not sure if I need to build the contents of the ADCodeGen folder or not, but cannot seem to get this folder to build.

alexliniger, 2023-04-25:
Sorry I know, I will add instructions and plan to add a docker container:

**These instructions are not tested, but is what I remember plus some old personal documentation**

### Installation:
Compared to the instructions for the other branches:
build blasfeo and hpipm into root with cmake following their changed instructions

CppAD:
```
Download CppAD
cd in folder
mkdir build
cd build
cmake ..
sudo make install
```
I think you can also use `sudo apt install cppad`

CppAD CodeGen
```
git clone https://github.com/joaoleal/CppADCodeGen.git CppADCodeGen
cd in folder
mkdir build
cd build
cmake ..
sudo make install
```

### Build:

Build CppAD Libs:
```
cd ADCodeGen
mkdir build
cd build
cmake ..
make
./ADCodeGen
```

Build MPCC:
```
mkdir build
cd build
cmake ..
make
./MPCC
```

ckurtz22, 2023-04-25:
Great, thank you!

ckurtz22, 2023-04-25:
#84 kinda fixes this as it adds the steps needed to the install script

alexliniger, 2023-04-27:
Thanks for updating the install instructions. I will test everything as soon as I have time and approve the merge requests.

closed 2023-04-27 by alexliniger