alexliniger/MPCC #86: Question about Track

KarasevPavel, 2023-05-02:
Hello, Alex!

I have a question about track for the fullsize model.

I work in the [fork](https://github.com/Bauman-Racing-Team/MPCC.git) of Your rep and already have implemented your **C++** version in  **Matlab**, using **Matlab** classes. 

There is a README in the rep, so You could easily check it out if You will have time.

My forked rep has some differences in car model (Other Pacejka tire model for lateral slip), new control input (brakes rate of change) and some other minor changes.

Also in my fork I have integrated Formula Student tracks from **FSSIM** and **eufs_sim**, but have problems with driving on them. MPCC works unpredictable and fails on the first lap corner on the **FSG** track.
 
On your track for the fullsize car MPCC works pretty great, so I have a question about requirements for the the track. 

What requirements should be taken into account, when we generate track bounds and track center line?

alexliniger, 2023-05-02:
The tunning of the weights can sometimes be a bit hard, when I added braking I have also seen that it makes the problem quite a bit harder. 

Some questions:

- Did you visualize the curvature of your tracks, if the curvature is very noisy it can result in issues and you may need to perform some smoothing, I normally use b-splines for this task.
- Are you using the fullsize branch with the "more correct" SQP implementation (the new branch now fixes one final strange detail in the fullsize branch where I used a mix of RK4 and matrix exponential discretization.

KarasevPavel, 2023-05-02:
@alexliniger Thank You for the quick response! 

1. I have used your resample algorithm from ArcLengthSpline.

With your algorithm I have resampled 94 points center line to 5000 points centerline.

Also I have visualized the final track center line after resampling, it looks quiet smooth.

If You have some free time, You could simply run my version in Matlab and watch the plotted center line (plotting of it is enabled by default).

2. I have used a **fullsize** branch for my fork.

What do You mean, when speaking about new branch? Is it a **fullsize** branch or **feat/cppad-codegen** branch?

alexliniger, 2023-05-03:
I have unfortunately no access to the Matlab  anymore. We have seen that resampling centerlines does not always work good. But for MPCC it is not as critical because the curvature only enters in the cost function and not in the dynamics. Generally resample with a b-spline including smoothing should work better.

With new branch I meant cppad-codegen, there you can also find parameters for a car with braking, but formula student tracks are always a bit special due to the tight boundaries. Maybe try smaller weights on the soft constraints (really small) and see if this helps. The soft constraints can make the problem unstable sometimes.

KarasevPavel, 2023-05-04:
Hello, Alex!

I have implemented your C++ codegen version of MPCC in my [**fork**](https://github.com/Bauman-Racing-Team/MPCC) with small changes of Cost values and model parameters. Practically all is the same. 

I have added some FS tracks from **FSSIM**and **eufs_sim** to the C++ implementation. Now all track **.json** files are stored in `Parameters/tracks` folder. More Information about it You could find in `C++/README.md` file in my [**fork**](https://github.com/Bauman-Racing-Team/MPCC).

Now in the rep there are two implementations of **FSG** track:
* Not presplined (**FSG.json**);
* Presplined (**FSG_nirajbasnet.json**) from **nirajbasnet** [**rep**](https://github.com/nirajbasnet/Nonlinear_MPCC_for_autonomous_racing).

On presplined version algorithm works better. You could easily check it (presplined FSG version is enabled by default).

But error status **(1)** appears very often, so I think that something works wrong, but I do not know where to look for the problem now.

Could You give me a quick explanation about methods to find such problems in MPС? May be there are some techniques for tuning MPC parameters and debugging (visualizing) MPC (dynamics/constrains) in the different horizon points?

Thank You in advance!