alexliniger/MPCC #87: How can I run the code in MATLAB

Vonct, 2023-05-31:
Hi，I was trying to run the code in MATLAB (based on Ubuntu20.04 & WSL2)

But I met some errors that were about calling hpipm solver, but I don't know what to do.
Hopefully, who run the code successfully can help me :)
## What did I do
1. I install BLASFEO & HPIPM by following the instructions on  `https://github.com/giaf/hpipm`
2. I also run `source env.sh` and then open matlab in the same terminal (I modified the env.sh file to set the PATH corret)
3. In addition, I add path to the matlab (using `set path`)

## error I met when I run simulation.m
```
Warning: MATLAB has disabled some advanced graphics rendering features by switching to software OpenGL. For
more information, click here. 
CVXInterface.m		      QuadProgInterface.m  augState.m	       getMPC_vars.m	 optimizer_mpcc.m
DiscretizedLinearizedModel.m  README.md		   borderAdjustment.m  getMPCmatrices.m  simulation.m
ObstacelsState.m	      SimTimeStep.m	   carBox.m	       getModelParams.m  splines
PlotLog.m		      Tracks		   env.sh	       getNewBorders.m	 unWrapX0.m
PlotPrediction.m	      YalmipInterface.m    findTheta.m	       hpipmInterface.m

Warning: The following error was caught while executing 'hpipm_ocp_qp_dim' class destructor:
Undefined function 'ocp_qp_dim_destroy' for input arguments of type 'double'.

Error in hpipm_ocp_qp_dim/delete (line 99)
			ocp_qp_dim_destroy(obj.C_dim);

Error in hpipm_ocp_qp_dim (line 45)
			if isa(in, 'char')

Error in hpipmInterface (line 39)
dims = hpipm_ocp_qp_dim(N);

Error in getMPCmatrices (line 81)
[X,U,dU,info] = hpipmInterface(stage,MPC_vars,ModelParams);

Error in optimizer_mpcc (line 29)
    [xpred, upred,dupred,info] = getMPCmatrices(TrackMPC.traj,MPC_vars,ModelParams,new_b,x,u,x0,u0);

Error in simulation (line 134)
    [x_up, u_up, b, exitflag,info] = optimizer_mpcc(TrackMPC,MPC_vars,ModelParams, n_cars, Y, x, u, x0,
    uprev); 
> In hpipm_ocp_qp_dim (line 45)
In hpipmInterface (line 39)
In getMPCmatrices (line 81)
In optimizer_mpcc (line 29)
In simulation (line 134) 
Unrecognized function or variable 'ocp_qp_dim_create'.

Error in hpipm_ocp_qp_dim (line 53)
				obj.C_dim = ocp_qp_dim_create(N);

Error in hpipmInterface (line 39)
dims = hpipm_ocp_qp_dim(N);

Error in getMPCmatrices (line 81)
[X,U,dU,info] = hpipmInterface(stage,MPC_vars,ModelParams);

Error in optimizer_mpcc (line 29)
    [xpred, upred,dupred,info] = getMPCmatrices(TrackMPC.traj,MPC_vars,ModelParams,new_b,x,u,x0,u0);

Error in simulation (line 134)
    [x_up, u_up, b, exitflag,info] = optimizer_mpcc(TrackMPC,MPC_vars,ModelParams, n_cars, Y, x, u, x0,
    uprev);
```

alexliniger, 2023-05-31:
Sorry I have no longer access to Matlab, but maybe @KarasevPavel can help. He seems to have a working updated fork of the matlab code.

https://github.com/Bauman-Racing-Team/MPCC

Vonct, 2023-05-31:
Thanks for your reply, I would ask if he know how to solver the problems!

Vonct, 2023-06-02:
Hi @KarasevPavel ！
Thanks for your help, now I can run the code successfully :)

alexliniger, 2023-06-02:
Thanks @KarasevPavel

closed 2023-06-02 by alexliniger