alexliniger/MPCC #90: Regarding the addition of dVs to the control input

kms8527, 2024-02-05:
[image]

Given the dynamics model, the control input dVs is used to increase \dot_{s}, yet s seems to have no mathematical correlation with the vehicle's speed (vx, vy). Thus, I suspect that dVs and \dot_{s} could not possibly affect the MPC performance, but in practice, it appears not to be the case. dVs is a virtual input that cannot be directly applied as a control input in real vehicles, and I'm unsure why Vs was introduced. Could you theoretically explain this?

alexliniger, 2024-02-05:
The connection comes via the lag error which forces `s` to be close to the projection of `X` and `Y` onto the reference path. Given that `s` is linked to the position `v_s` and `dv_s` are also linked to the position.

The result is similar to Frenet/Curvilinear coordinates but while keeping the Euclidean position.

fzhang327, 2024-10-29:
Thank you for your question, I also notice this.
Is this the arc-length dynamics described in "AMZ Driverless: The Full Autonomous Racing System"?

kms8527, 2024-10-29:
yeah, i think so.
you can also check "Optimization-based autonomous racing of 1:43 scale RC cars".