alexliniger/MPCC #93: Lyapunov Based Stability

prajwalthakur, 2024-03-07:
Hi, I wanted to talk about stability in mpcc . I've seen another similar question but curious on why the following two similar  papers talk about the stability but not the papers you have proposed .  
In my exploration, I've encountered few papers where authors have addressed stability (in similar application) in various ways, such as:

In the paper on MPCC by D. Lam [https://ieeexplore.ieee.org/document/5717042], the authors proposed the use of contractive constraints to ensure stability.

Similarly, in the paper referenced in the MPCC paper, 'Model Predictive Path-Following for Constrained Nonlinear Systems' [https://ieeexplore.ieee.org/document/5399744], a terminal penalty was imposed, demonstrating guaranteed closed-loop stability.

I'm curious why these types of stability guarantees have not been incorporated into your paper.? why its not neccessary?
Your perspectives on this would be highly valued.
thanks.

alexliniger, 2024-03-26:
Stability of the problem is quite hard to establish without going for trivial solutions, such as standing still at the end of the horizon. The issue is that computing recursive feasible sets and lyapunov functions for nonlinear control problems is extremely hard and in most cases not tractable. 

Trivial approaches like standing still (technically you even need to change the model to be well defined for standing still) are possible, but come with the cost that they are extremely conservative. Thus, either the car can only drive quite slow, or an extremely long horizon is needed. 

Thus, we went with the practical approach, of adding a reasonable terminal cost. Where the car should stay close to the reference path at the end of the horizon, giving the most room for errors. Additionally, we used a long prediction horizon to avoid some of the issues. 

I actually did quite a bit of work in this direction, with mainly working on recursive feasibility. Which guarantees that you never violate the constraints if you stay within this set. I think the two papers you mention where able to avoid this by not using state constraints. 

Here are two papers of mine working on recursive feasibility: 
https://arxiv.org/pdf/1701.08735.pdf
https://arxiv.org/pdf/2005.07691.pdf

prajwalthakur, 2024-03-28:
Hi, thanks for your comment.
The paper MPCC by D. Lam [https://ieeexplore.ieee.org/document/5717042] has state and input constraints.
"Which guarantees that you never violate the constraints if you stay within this set" is true, but only having this is not enough because ig this doesn't guarantee that the error will converge ,( the error may oscillate) .

alexliniger, 2024-04-02:
So this is all kind of specific to the autonomous racing application, there what you would like for "stability" is the convergence to a limit cycle (the racing trajectory), and the state should not really converge to an equilibrium. For other application you may want that you converge to an equilibrium, but the only one that really makes sense and is stable is kind of standing still. The issue is that you only know the path within the horizon and not byond, so you have 3 possibilities: 1. If you are standing still at the end of the horizon you don't really care what will be next. 2. You assume that you know the maximum curvature and you can use tools from robust control (however they will still not really work since the model is nonlinear). 3. You assume the road will be straight afterwards in this case things become simpler but still very hard due to the nonlinear dynamics. 

Generally, if you want to work on stability I would use frenet/curvilinear coordinates as I did here (https://arxiv.org/pdf/2005.07691.pdf) since this will make a lot of the steps easier. 

Regarding the convexification, consider this more a way to solve the NLP (sequential quadratic programming) than the fact that we have a linear system afterwards. The issue is that the system changes at every call to the slover, thus infinity horizon behavior (stability) can not really be studied using this model.

alexliniger, 2024-05-17:
I am closing the issue, you can re-open it if you have further questions

closed 2024-05-17 by alexliniger