alexliniger/MPCC #94: Inquire if GP based MPC code will be open-source

rogueyan5, 2024-04-04:
Hi Alex, 
may I ask if you plan to open source the relevant code for the model predictive controller based on Gaussian process regression designed in the paper "Learning Based Model Predictive Control for Autonomous Racing"? I am very interested in the deployment of Gaussian process regression in MPC.
Thanks!

alexliniger, 2024-04-04:
We currently do no plan to open source the code. It would also be quite hard to include it into the existing pipeline, since for the paper we used ForcesPro.

I think adding something similar to the pipeline would be possible, if you are ok to only use the offset form the GP but not the gradients. In this case you can just add a constant term into the vehicle dynamics as x_k+1 = f(x_k, u_k) + GP(x_k, u_k) and use the variance to compute the constraint tightening, similar to how we did it in this paper https://ieeexplore.ieee.org/document/7810413. In fact things would be very similar to this paper with the main difference that the uncertainty doesn't come from samples, but instead the samples are used to learn a GP, which is then used to compute the mean model uncertainty and the constraint tightening.

rogueyan5, 2024-04-04:
Thanks for the reply, I will try it according to the paper you gave.

alexliniger, 2024-05-17:
I am closing the issue, you can re-open it if you have further questions